Grubuntu/PieMenu #2: Odd shadows from "last cancelled pie"

pgilfernandez, 2024-01-15:
Hi,

In "press" trigger mode ("hover" mode is also affected by this issue but it is easier to analyze in "press" mode), if you cancel a pie, move the mouse pointer to another location and open the pie again, a "ghost" pie shows in your previous cancelled position showing just the shadows...

If you run a command from the pie, the pie closes and the "ghost" pie doesn't show.
BTW, I tested it under macOS:

https://github.com/Grubuntu/PieMenu-Improved/assets/5942369/9165cdb0-5f92-4b29-bacd-d4a072416f8f

Thanks!

```
OS: macOS 12.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35485 (Git)
Build type: Release
Branch: main
Hash: a662fbb2ff267fb8dd56fcc3d1ec55a5968b7f5a
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods:
  * PieMenu-Improved 1.2.7
  * SelectorToolbar
```

Grubuntu, 2024-01-16:
It seems that MacOs add shadows behind windows, 

We can try to disable this behavior with  : Qt.NoDropShadowWindowHint
[https://doc.qt.io/qtforpython-5/PySide2/QtCore/Qt.html#:~:text=Qt.NoDropShadowWindowHint](url)

https://github.com/Grubuntu/PieMenu-Improved/blob/198d2893364768d4ff27e2bce6f0bd2b8564a047/InitGui.py#L667

Try this instead :  
`self.menu.setWindowFlags(self.menu.windowFlags() | QtCore.Qt.FramelessWindowHint | Qt.NoDropShadowWindowHint)`

pgilfernandez, 2024-01-16:
Nice! It worked.

There is still a blink of the whole pie on the last position, it's like it shows up before getting the updated position. It repositions pretty fast but it's something to have a look at it as it could feels odd to some users.

Once again, this "glitch" is only present if you cancel the pie, if you run a command the pie opens in the right position.

Have a look at the video:

https://github.com/Grubuntu/PieMenu-Improved/assets/5942369/3a254088-410a-4e1d-b54b-86d06b333b53

Thanks!

Grubuntu, 2024-01-22:
> Once again, this "glitch" is only present if you cancel the pie, if you run a command the pie opens in the right position.
> 
> Have a look at the video:
> 

Today, I also noticed these glitches while running FreeCAD on an old and not powerful laptop. Maybe it's a display problem (graphics card)?

pgilfernandez, 2024-01-23:
I don't know, my computer is not new but it's pretty fast...

Grubuntu, 2024-01-27:
Hi @pgilfernandez 

I think I found where the problem is:

in the definition 
`def updateCommands(keyValue=None, context=False):`

```
                      try:
                            Gui.activateWorkbench(i + "Workbench")
                        except:
                            None

                Gui.activateWorkbench(lastWorkbench.__class__.__name__)
```
(We find this pattern elsewhere in the code 'def buttonList')

These lines reload lastworkbench each time we activate a Piemenu, this is what causes ghosting I think.
You could try to comment these lines to do some testing.

I will try to find a workaround, but I have no idea to avoid this behavior without risking causing problems when loading a toolbar from a WB that is not loaded... Something to think about..

pgilfernandez, 2024-01-27:
I'll test it when I can, I haven't experienced with behavior on a while so I should study it deeply.
I'm busy now with some other features or tweaks that, under my opinion, are more necessary and exciting ;)

Grubuntu, 2024-01-28:
1.3.6 : Fixes the problem (need some tests)

closed 2024-01-28 by Grubuntu