Grubuntu/PieMenu #3: [Feature request] Search box for Tools commands

pgilfernandez, 2024-01-15:
Hi,

It would be so handy being able to filter the looooong commands list from "Tools" with a search box.

Thanks!!

Grubuntu, 2024-01-15:
Yes, i think it seems feasible and would be very practical.

Grubuntu, 2024-01-16:
I have merge the new version 1.3.1 : With search box 

(I thought it would have been much more complicated to do than that !)

[image]

pgilfernandez, 2024-01-16:
What about a final "simple" request?: how about adding an "x" icon to clean the filter?

Thanks!

Grubuntu, 2024-01-16:
I have updated the code with an icon for clear search box but I'm not very satisfied about the icon.
[image]

pgilfernandez, 2024-01-16:
Great!

Don't worry, tomorrow I design one that better fits =)

closed 2024-01-17 by Grubuntu