Grubuntu/PieMenu #4: [Feature request] Subpies!

pgilfernandez, 2024-01-15:
Hi,

I don't know how hard would be this feature to code, but it would be cool to see it implemented: a pie command that opens another pie, this way you could have a master pie that runs normal commands or "expand" more commands by running a "folder" command that links to the other pie... did I explained myself correctly?

Thanks!

Grubuntu, 2024-01-15:
I understand what you want. It requires menu nesting. Something to think about

Grubuntu, 2024-01-16:
Fun  workaround until I have time to code this. You create a macro that contains:

```
from PySide import QtGui, QtCore

def simulate_key_press():
    key_event = QtGui.QKeyEvent(QtCore.QEvent.KeyPress, QtCore.Qt.Key_W, QtCore.Qt.NoModifier)
    QtGui.QApplication.postEvent(FreeCADGui.getMainWindow(), key_event)
simulate_key_press()
```

it simulate the press on the key W here.

Add this macro to a toolbar via Custom toolbar in Freecad and you can finally add this macro in a PieMenu
[image]

Set the shortcut corresponding to the PieMenu you want to open with the macro
[image]

 "et voilà "

[image]

pgilfernandez, 2024-01-16:
Yes!!

Awesome, I also thought about this workaround but I didn't know how to code the macro... thanks for sharing it!

I already have it working and I'm gonna use it everyday  =D

Grubuntu, 2024-01-16:
```
from PySide import QtGui, QtCore

def simulate_key_press():
    # Create a key press event for the "Q" key
    key_event_q1 = QtGui.QKeyEvent(QtCore.QEvent.KeyPress, QtCore.Qt.Key_Q, QtCore.Qt.NoModifier)
    
    # Create a second key press event for the "Q" key
    key_event_q2 = QtGui.QKeyEvent(QtCore.QEvent.KeyPress, QtCore.Qt.Key_Q, QtCore.Qt.NoModifier)

    # Send the events to the FreeCAD application
    QtGui.QApplication.postEvent(FreeCADGui.getMainWindow(), key_event_q1)
    QtGui.QApplication.postEvent(FreeCADGui.getMainWindow(), key_event_q2)

# Call the function to simulate pressing the "Q" key twice
simulate_key_press()
```

You're welcome.

pgilfernandez, 2024-01-16:
Awesome again!
Thanks!!

Grubuntu, 2024-06-03:
Fix in version 1.6