Grubuntu/PieMenu #9: [Feature request] theme stylesheets

pgilfernandez, 2024-01-16:
As it happens in FreeCAD, the styles are stored separately to the code into QSS stylesheet files. I've seen you added an additional "transparent" style which is nice but...

It would be nice to save the button styles in files stored in `/Resources/Stylesheets/`, change the QCheckBox for a QComboBox that reads that folder and selects as active the one you desire. In that way it would be so easy to extend new styles that better suit dark/light themes or even those ones created with accessibility in mind.

Now you would be thinking: and what about the styles that aren't part of the pie buttons, that is, styleMenuClose, styleContainer, styleCombo, styleQuickMenu and styleQuickMenuItem?

Well, the idea would be adding them to the QSS stylesheet as well and instead of passing the code to the desired QWidget , creating an ID to that object and using it later in the stylesheet. For instance, now:
```
styleMenuClose = ("""
        QToolButton {
            background-color: rgba(60,60,60,255);
            color: silver;
            border: 1px solid #1e1e1e;
        }

        QToolButton::menu-indicator {
            image: none;
        }

        """)
```
And injected here:
```
button = QtGui.QToolButton()
…
button.setStyleSheet(styleMenuClose + radius)
```
The idea would be to create a stylesheet that has this code:
```
        QToolButton#styleMenuClose {
            background-color: rgba(60,60,60,255);
            color: silver;
            border: 1px solid #1e1e1e;
        }

        QToolButton#styleMenuClose::menu-indicator {
            image: none;
        }
```
And create the object name so that it connects with the QWidwet as:
```
button = QtGui.QToolButton()
…
button.setObjectName("styleMenuClose")
```

What do you think?

And, indeed, I would take care of creating nice stylesheets as I’m good at that ;)

Grubuntu, 2024-01-17:
Yes, it's a great idea.

You are right, moreover, by separating the style sheets, the code will be more readable.

Grubuntu, 2024-01-17:
Hi

Thanks to your advice, I made progress on managing style sheets. Here is the draft where I am.

[PieMenu.zip](https://github.com/Grubuntu/PieMenu-Improved/files/13966550/PieMenu.zip)

[image]

pgilfernandez, 2024-01-17:
Great!
I had a loot at it and works very good.

I have some questions:
- is the code finished? I mean, do you want to add anything else?
- I can start working on styles from this point and add some new styles

And some suggestions:
- instead of a `/styles/` folder, I would say it is better to name it `/Stylesheets/`, as FreeCAD does in the main app
- the same way, Qt uses `.qss`as the extension for their stylesheets (and not `.css` as in web design)
- finally, if you don't want to release this as it is still a draft, we can work in a branch, this way I would be able to make Pull Requests... but, as you want

Grubuntu, 2024-01-17:
> * is the code finished? I mean, do you want to add anything else?
It is never finished ;)   But I think we can stop major developments for the moment.
> * I can start working on styles from this point and add some new styles
Yes of course !

> And some suggestions:
> 
> * instead of a `/styles/` folder, I would say it is better to name it `/Stylesheets/`, as FreeCAD does in the main app
> * the same way, Qt uses `.qss`as the extension for their stylesheets (and not `.css` as in web design)
I will modify that

> * I  found a few issues in your .css styles, do you want me to fix them and attach them to this conversation or
> * finally, 

It goes without saying, I'm not a good coder, so thank you in advance for correcting me.

>if you don't want to release this as it is still a draft, we can work in a branch, this way I would be able to make Pull Requests... but, as you want

Sure, that's a very good idea

Grubuntu, 2024-01-17:
This is the branch (sorry I'm not yet very comfortable with github)

https://github.com/Grubuntu/PieMenu-Improved/tree/stylesheets-dev

pgilfernandez, 2024-01-17:
> It goes without saying, I'm not a good coder, so thank you in advance for correcting me.

Hahaha, don't worry, I'm neither a good coder or coder, I just know to do basic things and learn by reading others code and copy-pasting...
I'm truly pro at CSS and have pretty much experience with FreeCAD QSS as I expend lots of hours creating the first stylesheets, so that's the part I'm better on.

I think we are (mostly you) improving the plugin a lot making it so nice.
There still work to do to reach which is for me the best pie plugin one: realthunder's LinkStage branch one, but we are not far and his ones doesn't work in FreeCAD normal one so we are there ;)

Tomorrow I will start creating PRs on the branch, thanks.

Grubuntu, 2024-01-18:
You probably notice it : we need to apply "radius" before "styleCurrentTheme", otherwise it doesn't make the radius define in 'def radiusSize'

`button.setStyleSheet(styleCurrentTheme + radius)`  ==>

`button.setStyleSheet(radius + styleCurrentTheme)`

unless you have another solution?

pgilfernandez, 2024-01-18:
> You probably notice it : we need to apply "radius" before "styleCurrentTheme", otherwise it doesn't make the radius define in 'def radiusSize'
> 
> `button.setStyleSheet(styleCurrentTheme + radius)` ==>
> 
> `button.setStyleSheet(radius + styleCurrentTheme)`
> 
> unless you have another idea?

I would say the right way is to apply first "styleCurrentTheme" and then "radius" as this way the last one is the one that overrides the previous styles and set the right radius... anyway let me test it tomorrow as today I didn't have time. But it looks fine to be and a matter to polish the styles and create some new ones in the way (for me the fun part) ;)

Grubuntu, 2024-01-27:
v1.3.5 : The new styles you created are in the code, we can continue to add more.
Thanks for your work.

closed 2024-01-27 by Grubuntu