Grubuntu/PieMenu #14: Shape of the piemenu when there are less than 8 elements

Grubuntu, 2024-01-19:
When a pieMenu has less than 6 to 8 elements, the buttons are far from each other.
Actually :
[image]
[image]
[image]
[image]
[image]

I thought I modified the code a little to ensure that when there are less than 8 elements in the menu, they are positioned first at the top... in the shape of a rainbow. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Do you think this is a good idea, or is this not a good design for use?

Actual code :
https://github.com/Grubuntu/PieMenu-Improved/blob/56a110cf3a77ed17c2eed5a9fb1e663d03f26295/InitGui.py#L821-L832

Proposed code change :

```
            if commandNumber == 1:
                angle = 0
                buttonSize = self.buttonSize
                #####Adjust PieMenus when there is less than 8 QtoolButtons  ######
                angleStart = 3 * math.pi / 2 - angle
            else:
                if commandNumber < 8:
                    angle = 2 * math.pi / 8
                else:
                    angle = 2 * math.pi / commandNumber
                    
                buttonRadius = math.sin(angle / 2) * self.radius
                buttonSize = math.trunc(2 * buttonRadius / math.sqrt(2))
            if commandNumber > 1:
                angleStart = 3 * math.pi / 2 - (angle*(commandNumber+1))/2
                ##### End Adjust PieMenus when there is less than 8 QtoolButtons######
            if buttonSize > self.buttonSize:
                buttonSize = self.buttonSize
            else:
                pass
```

pgilfernandez, 2024-01-21:
> Do you think this is a good idea, or is this not a good design for use?

Yes, I like the idea as I know some users would prefer this behavior but I would like to propose you something bigger and more customizable: some users reported they would like pies to not open in a circular shape but as a stack of table of rows and columns.
That way, I would suggest to extend each pie with a new parameter that let users choose the "type of pie". That way we can add different ways of arranging the pie commands and users just choose them from a list.
Some ideas:

[image]

[image]

[image]

The same way you can imagine another set vertically.

What do you think?

Grubuntu, 2024-01-21:
> 
> * circle (classic)
> * circle (top to bottom, the one you are proposing)
> * circle (bottom to top, why not?)
> * table top
> * table bottom
> * table top-bottom
> * table left
> * table right
> * table left-right
> * ... more?
> 
> What do you think?

These are very good ideas, as long as the user can choose, it's very good.

This will take a little time to complete and make some choices (in particular, for the maximum number of shortcuts to display). Great work to come.

pgilfernandez, 2024-01-21:
Well, I think common sense will tell each user what maximum number of commands to add, but I guess it has to be around 12 on each side (in the case you use top-and-bottom or left-and-right)... BTW, I don't like using more than 8 commands in circle ones as it's the only way to truly remember their position.

What I'm unsure about, is if we should let users decide their number of rows and columns or just play with fixed ones we decide internally and see if it works or there are users that ask for further customization...

Related with your original suggestion, I would like to ask for another "simple" feature that makes this one a little bit more complicated, just because it adds a new dimension on how to arrange icons, but let me introduce it in a new "Feature request", [here it is](https://github.com/Grubuntu/PieMenu-Improved/issues/16).

Grubuntu, 2024-01-27:
v1.3.5 : I have updated the code with some shapes.
We will be able to add others as the tests progress.

closed 2024-01-27 by Grubuntu