Grubuntu/PieMenu #16: [Feature request] Command names in pie items

pgilfernandez, 2024-01-21:
I think it has been already suggested or requested somewhere in FreeCAD forums, and some previous attempts did it (realthunder ones has it, the following screenshot comes from his LinkStage branch): **add the option to users to see the command names in the pies**:

[image]

Sometimes command icons might be confusion and not clear but their names help to choose them correctly.

Once again, to keep it flexible to customization, I would suggest that, if implemented, it is added as "per pie" option, so you could have some pies with icons and another ones with icons + names. For instance, circle pies are pretty nice to have names but the ones we were talking about that get arranged in a table (rows and columns) may not be that handy unless you only have a column... (well, it's something to be seen)

Grubuntu, 2024-01-22:
I saw that this was a requested feature.
I'll have to look at how it's implemented in the Link.

Grubuntu, 2024-01-27:
v1.3.5  : I have updated the code with this new feature.
Thanks for the idea !

closed 2024-01-27 by Grubuntu