Grubuntu/PieMenu #20: Pies with command names

pgilfernandez, 2024-01-27:
Very cool your implementation! it looks so promising. The core work is done, I really like it!
I have some comments to make them more robust and prettier:

### Allowance
Now you are allowing them only in circular but the would fit in Rainbow and "left and right" types:
This would be a just icons option:
[image]

Then you can add names in the outer direction:
[image]

This technique also works with Rainbow pies:
[image]

### Items arrangement in circle pies
As the feature is already there (and working so nicely), I would say we just need to work a little bit more in the display/layout maths.
If this is a normal pie with 8 icons:
[image]

This is the pie with command names activated:
[image]

And this is the real example:
[image]

As you see, they get need more spacing between them. In a first approach this would be what I suggest:
1) Vertical spacing so that the names have space to not overlap vertically:
[image]

2) Then adding some maths to that top and bottom items get nicely aligned:
[image]

3) at that point you will see that the "circle" arranging gets lost (because of the manual vertical alignment):
[image]

4) so that we would need to arrange them back in the following way:
[image]

If you can add maths to that arrangement it would be awesome.

pgilfernandez, 2024-02-09:
> I would like to let you know that I'm working on command names for LeftRight pie shape, just in case you haven't started working on it ;) this one: <img alt="300193975-02266510-f4ac-4e16-95e0-7aca52ceb499" width="475" src="https://private-user-images.githubusercontent.com/5942369/300902814-775fc40e-cfdf-460b-bfb7-80d5c43ab8fe.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc0OTYzNzAsIm5iZiI6MTcwNzQ5NjA3MCwicGF0aCI6Ii81OTQyMzY5LzMwMDkwMjgxNC03NzVmYzQwZS1jZmRmLTQ2MGItYmZiNy04MGQ1YzQzYWI4ZmUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDlUMTYyNzUwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTIyM2E5N2NmZjRlMWZjMDcyNmM0ZDcxOWViNTU1ODIwZTQ3ZGFmY2UyY2MxYjRmYzNhMGRjOWZiZDJmM2YzNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.OZqmE-0rKP0ttQ9kUyfZhf7GbD6l3q5x7F_pIz_L3S0">

This feature has already been merged 7cc7920e0344667239dcbb7099a8588003acd40e
=)

Grubuntu, 2024-02-15:
I modified the Pie shape with the commands activated for better spacing management:
[image]
[image]

it's in development version 139 to test: https://github.com/Grubuntu/PieMenu/tree/next-dev139

pgilfernandez, 2024-02-15:
It looks like we've been working on the same feature in parallel, hehehe.
I feel like my algorithm has a better layout but it still doesn't work properly depending on the number of commands:
[image]

So, if you don't mind, I'm gonna try to modify your layout to match mine. I have now a spare hour, let's see if it's enough to finish it ;)

pgilfernandez, 2024-02-15:
I've PR the branch with my proposal for X distribution: https://github.com/Grubuntu/PieMenu/pull/47

Your Y one is different to mine as mine adds a spacing between icons, would you mind to include it as you know better your algorithm than me? I used a spacing of `buttonSize * 1.2` in case you want to use that one.

BTW, our both approaches were pretty similar in concept but your implementation is much better (that's why mine wasn't still working with any number of icons...)

Grubuntu, 2024-02-15:
Your algorithm is more beautiful than mine. I failed miserably with trigonometry ;)

>  I used a spacing of buttonSize * 1.2 in case you want to use that one.

I fixed minimum Y spacing in this commit : https://github.com/Grubuntu/PieMenu/commit/14f136b20166563c3804c50c2e062d2577583d5a

pgilfernandez, 2024-02-15:
It looks great! Good team work, hehehe.

Tomorrow or this weekend I would like to fine tune the stylesheets and make the new CommandNames styleable, let me PR it to the new-139 branch, ok?

Grubuntu, 2024-02-16:
Of course.

closed 2024-02-29 by Grubuntu