Grubuntu/PieMenu #23: Custom pie menu depending of the workbench

Grubuntu, 2024-01-29:
I started working on the modification request (https://forum.freecad.org/viewtopic.php?t=84101&start=70#p736859) to be able to define a Workbench at each PieMenu, this one opening by default in the Workbench in question.

I created a specific branch :

https://github.com/Grubuntu/PieMenu/tree/workbench-dependant

Do not hesitate to test and report problems and possible improvements.

[image]

pgilfernandez, 2024-01-29:
Hi,

I had this feature also in my mind as I have seen it in realthunder's branch and it's so handy, nice to see you have started working on it, people are gonna love it (me included, hehehe).

BTW, don't forget to add a "Global" fake workbench option that means you want that pie to be active everywhere ;)

Tomorrow I'll try the branch and let you know
=)

pgilfernandez, 2024-01-30:
I gave it a try and it doesn't work for me... If I set a specific workbench to a pie it is visible in all of them.

I also noticed:
- in the workbench list there is a "None" entry which is no desiderable unless it means "all" or "global"
- if "None" means "disabled", then I would rename it this way and add the "global" option

Grubuntu, 2024-01-30:
> I gave it a try and it doesn't work for me... If I set a specific workbench to a pie it is visible in all of them.

The desired behavior is that the "global" shortcut (TAB or other global shortcut assigned) triggers the PieMenu assigned to the Workbench. The custom shortcut (if there is one) remains functional in all workbenches. 

You can open a Piemenu in different ways, some having priority over others:
Press Global Shortcut > 
IF (Context checked) AND (Context OK (Vertex, Faces selected etc)) > OPEN PieMenu with the context corresponding 
ELSE IF : (WB Assigned actived) AND (PieMenu with assigned WB exist) > OPEN : PieMenu assigned to the WB
ELSE  IF : (ToolBar checked) OPEN : ToolBar selected in QuickMenu (which is checked in QuickMenu)
ELSE : OPEN Default Pie (which is checked in QuickMenu, hence my reluctance to hide this menu)

Press Custom Shortcut > Open PieMenu assigned to this custom shortcut (no matter the context or the WB)

> I also noticed:
> 
> * in the workbench list there is a "None" entry which is no desiderable unless it means "all" or "global"

"None" should mean "No Workbench assigned to this PieMenu"

> * if "None" means "disabled", then I would rename it this way and add the "global" option
> 
> Anyway, I guess you are still working on this feature, I'll stay tuned ;)

closed 2024-03-16 by Grubuntu