Grubuntu/PieMenu #26: Disable Hover mode for some "Table pies"

pgilfernandez, 2024-01-29:
When you setup a pie with more than 1 row/column, the "hover" pie mode turns unusable...

I would suggest to move this option as a "per pie" so that we could disable it when this "table pie with more than 1 row/column" happens and that way it turns more flexible as you could have some pies with hover and some with press.

What do you think?

Grubuntu, 2024-01-30:
I don't have the problem on my side. Another MacOS difficulty?

pgilfernandez, 2024-01-30:
It's not because of any macOS difficulty but the concept itself, with more or less trigger delay, what you want is to trigger it fast. If you need to "jump" a row of icons to reach the one you want to trigger, there is a chance the delay gets too short or long... that's why it is better to relay on press than to have to "calculate" that trigger delay.

pgilfernandez, 2024-01-30:
Having the option to configure it "per pie" gives you the flexibility to tweak it as needed for that specific pie... Another option would be to move the trigger delay also to the pie settings and you can set different delays depending on you pie.

Maybe, the best and more flexible option would be to move both, the press/hover mode and it's delay to "per pie"...

Grubuntu, 2024-01-30:
This also means modifying the Quick Menu.  Add more additional parameters in the config file.  I don't believe that the user switches from Hover mode to Press mode depending on each menu bar, it is very uncomfortable to have different behavior between each PieMenu.

pgilfernandez, 2024-01-30:
Well, it's just a matter of letting users decide what they want "per pie", because the layout and behavior of a circular pie doesn't have to match the one you want for a table one...
It would add more flexibility but, as you said, more complexity to the pie configuration.

As medium to pro user I would vote for more flexibility and options "per pie", but it's just an opinion, hehehe

pgilfernandez, 2024-03-01:
With the new shapes added (very cool ones btw), I would think it's the moment to have a look at this feature, I still think we should move the trigger mode (hover or push) to be a per pie option but, if you don't agree we can go with another approach, disable "hover mode" when you are using some kind of shapes, that is:

1. it works only if you have "hover mode" enabled
2. "hover mode" disables automatically when using any of this modes because they are incompatible with "2 or more rows of commands"): concentric, star, table (all)

What do you think?

Grubuntu, 2024-03-01:
I've already tried to do this and so far I've been unsuccessful.
I haven't managed to find the right approach with mouse and keyboard event capture yet.

pgilfernandez, 2024-03-01:
Umm, I thought it was gonna be something pretty easy... at least with the simple way I told you... I see I'm wrong.
Keep me informed of further trials, please

Grubuntu, 2024-03-06:
I've finally managed to incorporate the change to the trigger mode for each PieMenu.

[image]

Give it a try, let me know.
Thanks

pgilfernandez, 2024-03-08:
Hi,
I'm pretty busy lately with real life but I took a break a few minutes ago to have a look at this new feature: it works like a charm and I love it.
I have some pies that are meant with just 8 items in a "pie shape" that are perfect to be triggered on hover as they are way faster to be used, but others that have lots of items that need to be triggered with a click so I think this feature is gorgeous.

Thanks!!

closed 2024-03-16 by Grubuntu