Grubuntu/PieMenu #34: SIGSEGV when creating a new pie

hasecilu, 2024-01-31:
Since I had some parameter from mdkus version I decided to start from zero deleting PieMenu parameters.

Now when I try to create a new menu when switching to it in the GUI to add the commands FreeCAD crashes both when using new menu and copy current menu. In the parameters there is no update on IndexList.

Maybe we would need to enable some kind of console debugging with: `FreeCAD.Console.PrintMessage()`.

Could you try to create a new menu?
Tested on 0.21.2 and 0.22.dev versions.
```
Cannot find icon: /home/user/.cache/FreeCAD/Cache/AddonManager/MacroIcons/Macro_CenterOfMass.png
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3e710) [0x7d1c5025c710]
#1  /usr/lib/libQt5Core.so.5(+0x39bcb0) [0x7d1c50b9bcb0]
```

Grubuntu, 2024-01-31:
No problem noted here on a new installation (new user.cfg file). I tried on 0.21 and 0.22.

Do you have the 3 menus created by default during a new installation?
[image]

hasecilu, 2024-01-31:
Yes, they work fine, only I noted that (maybe because of the crash) values for pie and button sizes resets to default. 

Just when I click on "WBs" the crahs happens.

I'm on Arch Linux
[image]

Grubuntu, 2024-01-31:
Try to write in the parameters via command line to see if it is not a write permission problem:

```
paramPath = "User parameter:BaseApp/PieMenu"
paramGet = App.ParamGet(paramPath)
paramGet.SetString("Test", "Test_Ok")
```

After, check with Tools >Edit parmeters... to check if the string "Test_Ok" was written well
[image]

restart Freecad and  check again that the value is still present in the parameters.

hasecilu, 2024-01-31:
That's not the problem, it appears. In W10 seems to no be problems.
Could be something Linux related or something specific to my system. Maybe if someone could test on other Linux systems. I'll try previous versions.

hasecilu, 2024-01-31:
Up to 70d58a83 seems to work reliable fine.
Definitiely on 50b8533f there are crashes

Grubuntu, 2024-02-01:
Try to comment each lines contains and see if it works again:

```
comboShape.addItem(" ") # Add separator
comboShape.model().item(comboShape.count()-1).setEnabled(False)  # Disable the separator 
```

PR #35 (2024-02-01): bugfix: Revert addition of separators on `comboShape`
At least, on Arch Linux the addition of separators caused crashes.
Problem seems to not occur on Mac and Windows machines.
Don't know about other Linux/BSD distributions.

Unfortunately this PR removes the separators, that is a good idea.

Fix #34
commit b55bf771 (2024-02-01): bugfix: Revert addition of separators on `comboShape`

At least, on Arch Linux the addition of separators caused crashes.
Problem seems to not occur on Mac and Windows machines.
Don't know about other Linux/BSD distributions.

Fix #34
closed 2024-02-01 by Grubuntu