Grubuntu/PieMenu #36: Update `package.xml` file

hasecilu, 2024-02-01:
Now that https://github.com/FreeCAD/FreeCAD-addons/pull/285 was merged the Addon manager redirects to this repo but data is not updated

Grubuntu, 2024-02-01:
It doesn't work, Freecad add a directory "Piemenu-main",  where did I go wrong ?
[image]

hasecilu, 2024-02-01:
Change to `main` branch
`  <url type="repository" branch="main">https://github.com/Grubuntu/PieMenu</url>`

Grubuntu, 2024-02-01:
Solved : rename the main branch to master and edit the package.xml.
Thank you for your help you put me on the right way.

hasecilu, 2024-02-01:
It wasn't easier to only edit the xml?

Grubuntu, 2024-02-01:
It doesn't work with branch name 'main', it seems FreeCad expected a 'master' branch.

hasecilu, 2024-02-01:
Mmm, I checked it and it's because in this file I didn't include the branch :sweat_smile: .
https://github.com/FreeCAD/FreeCAD-addons/blob/master/.gitmodules

closed 2024-02-01 by Grubuntu