Grubuntu/PieMenu #38: help for translation

Grubuntu, 2024-02-01:
I'm trying to make the translation files in French: I managed to generate the .ts file, and translate with Qt Linguist. I generate the .qm file and drop the 2 files in the translation folder of PieMenu and if I set the language of FreeCAD to French, but the translation does not take place in PieMenu. I screwed up somewhere again ?

[translationFR.zip](https://github.com/Grubuntu/PieMenu/files/14129925/translationFR.zip)

hasecilu, 2024-02-02:
Hi, I'm checking your file and I think it could be because of the relative paths
```
    <message>
        <location filename="../../InitGui.py" line="3067"/>
        <source>Invalid shortcut ! Current global shortcut : </source>
        <translation>¡Atajo inválido! Atajo global actual: </translation>
    </message>
```
French
```
    <message>
        <location filename="../../../AppData/Roaming/FreeCAD/Mod/PieMenu/InitGui.py" line="3146"/>
        <source>Invalid shortcut ! Current global shortcut : </source>
        <translation>Raccourci invalide ! Raccourci global actuel : </translation>
    </message>
```

Grubuntu, 2024-02-02:
Great, Thank you very much it works.

closed 2024-02-02 by Grubuntu