Grubuntu/PieMenu #40: To test : Added checkbox for Pocket, changing spinbox type to accept calculations

Grubuntu, 2024-02-03:
Hi

I have integrated some new features into the branch [workbench-dependant](https://github.com/Grubuntu/PieMenu/tree/workbench-dependant), if you can test it on your side.

Added a checkbox for Pocket : Through all, Symmetric, Reversed

[image]

Changing spinbox type to accept calculations 
Added units in spinbox

Fix some weird behavior with Context (customshortcut was not a priority compared to the context and others things...)
Reactivation of the quickmenu in the sketcher, the problem seems resolved

Thanks

hasecilu, 2024-02-05:
Hi, I have tested it and all new features seems to work fine.
The only problem I see is that the text is clipped

[image]

Grubuntu, 2024-02-05:
Thank you for your help.

We will try to find a solution to keep the checkboxes aligned and not clip the words if possible...
I will try with style modifications.

Grubuntu, 2024-02-06:
I made some changes, I hope it will be better this way. 

I fixed width of the checkboxes at 220px, I hope it will be enough, and moved in X the checkboxes to keep the alignment with the spinbox.
Added  style in the CSSs to fix the margin-left (but we can modify other style if we want.)

[image]

I found a bug on 0.21, there is not the attribute I use to retrieve current units (Gui.ActiveDocument.Document.UnitSystem) so I add a "try" and avoid to use Units in spinbox in 0.21 at this time, I will search for a best solution.

hasecilu, 2024-02-06:
Hi, I tested again and the alignment is working fine, thanks!

pgilfernandez, 2024-02-07:
Hi,

I finally had time to test it and I found a problem:

- when I try to create a fillet/chamfer the spinbox starts with 1000,00 mm
- if I try to pad/pocket now it starts with 1Km

There is something wrong with the units... at least here under macOS

Grubuntu, 2024-02-07:
> Hi,
> 
> I finally had time to test it and I found a problem:
> 
> * when I try to create a fillet/chamfer the spinbox starts with 1000,00 mm
> * if I try to pad/pocket now it starts with 1Km
> 
> There is something wrong with the units... at least here under macOS

ha damn. What do this return to you?

```
unit_schema = FreeCAD.ParamGet("User parameter:BaseApp/Preferences/Units").GetInt("UserSchema")
unit_schema = FreeCAD.Units.listSchemas(unit_schema)
print(unit_schema)
```

pgilfernandez, 2024-02-07:
> I find it, I made a mistake on the line 1061 : (float(float(

It fixed the fillet/chamfer bug but pad/pocket still starts with 1km lenght...

hasecilu, 2024-02-08:
For me it appears the 10mm and 5mm defaults for pad/pocket.

Somehow I concentrated so much in the checkboxes that I missed that the spinbox it's a little short. Maybe could be about 3-5 char-width bigger. See previous images, `mm` is not seen.

Grubuntu, 2024-02-08:
> > I find it, I made a mistake on the line 1061 : (float(float(
> 
> It fixed the fillet/chamfer bug but pad/pocket still starts with 1km lenght...

Thanks for your tests.

I just made some modifications, if you can test again and I added some debug messages in the console to see the value taken by the spinbox. Thank you for your feedback.

@hasecilu : yes indeed, I think it depends on the user's screen resolution, we will have to take this into account, perhaps by adjusting the styles.

pgilfernandez, 2024-02-08:
fillet/chamfer is still not working:
```
Radius quantity : 1000.0 mm
```
but it was supposed to be 1mm (that's what the task panel spinbox reads as default radius).

The same with pad/pocket:
```
Pocket or Pad quantity : 10000.0 mm
```
while the default length is 10mm

Grubuntu, 2024-02-08:
> fillet/chamfer is still not working:

Can you try to add a line to initialize the variable 'quantity' before use and add some print debug to see the value of g.Object.Radius and other g.Object.

                        
```

      layoutMidPlane = QHBoxLayout()
                        layoutReversed = QHBoxLayout()
                        layoutThroughAll = QHBoxLayout()
                        
                        quantity = Units.Quantity('1 mm')

                        if (str(fonctionActive) == '<PartDesign::Fillet>'):
                            quantity = Units.Quantity("{} {}".format(float(g.Object.Radius), unit))

                            print("g.Object.Radius:", g.Object.Radius)

                            print("Radius quantity :", quantity)
                            # self.double_spinbox.setProperty('value', quantity)
                        elif (str(fonctionActive) == '<PartDesign::Chamfer>'):
                            quantity = Units.Quantity("{} {}".format(float(g.Object.Size), unit))
```

There is a problem with the interpretation of the values but I have no clue at the moment.

I'm going to try installing a MacOs VM on my machine to be able to test locally.

Grubuntu, 2024-02-09:
@pgilfernandez 

After a few hours of modifying different parameters I finally managed to reproduce the problem in my MacOS VM:

[image]

It seems that setting number separator to "None" and Decimal to "Comma" in locale in MacOS is causing problem for FreeCAD. 

[image]

You must then either put a different number separator in the MacOS settings, or modify the FreeCAD preferences and set the "Number Format" to another value : "Selected Language" and "C/POSIX" seem to work correctly...

[image]

I don't know if this is questionable behavior of my code in PieMenu or FreeCAD code.

Can you try and tell me if this solves the problem. Unfortunately, I don't know how else to work around the problem at the moment. THANKS

pgilfernandez, 2024-02-09:
Thanks for the tests.
You pointed out more or less where the problem was caused:
- the problem is not the separator (comma or dot)
- but the "Default unit system" you set

[image]

Mine is `MKS (m, kg, s, º)` while yours is `Standard (mm, kg, s, º)`. If I set the Standard one then it works fine, but the problem is still there as, what I think it's happening is that the spinbox is multiplying 1 units of the "default unit system" by the default fillet/chamfer value.
That is:
- if you set MKS: 1unit (1m = 1000mm) * 1mm = 1000mm
- if you set Standard: 1unit (1mm) * 1mm = 1mm

This theory is double validated:
1) if you set a fillet/chamfer to 2mm (or any other value) using the Task panel spinbox (the normal one) and the open your pie and run the fillet/chamfer command, it opens one fast spinbox and then you read as the first value 2000mm.

2) if you set your "Default unit system" to Imperial or any other one, when you open the pie spinbox you will see that the starting value is 1 unit converted into mm.

Did I explain myself correctly?

Grubuntu, 2024-02-09:
Yes your explanations are clear, thank you. You are right, it seems that Freecad only works in mm internally.

I'll have to handle unit conversions... unless there's another mechanism to handle this directly in Freecad... I'll dig into it...

Grubuntu, 2024-02-09:
```
print("g.Object.Radius:", g.Object.Radius)
print("Radius quantity :", quantity)
```

[image]

Grubuntu, 2024-02-10:
I updated the code, it works on Windows and seems to work on my MacOS VM.

Unfortunately, I have a problem with my MacOS VM: I made a short video to show the problem (I had uninstalled PieMenu to make the videos). The FreeCAD spinbox behaves strangely depending on certain MacOS and FreeCAD settings. And when PieMEnu is installed we find the same strange behavior in the PieMenu spinbox. I assume this is coming from my VM.

https://youtu.be/eg2Vb1I4igA

pgilfernandez, 2024-02-10:
> I updated the code, it works on Windows and seems to work on my MacOS VM.

Great, not it works perfectly.

> Unfortunately, I have a problem with my MacOS VM: I made a short video to show the problem (I had uninstalled PieMenu to make the videos). The FreeCAD spinbox behaves strangely depending on certain [...]

I confirm that behavior also in normal macOS (I mean not virtualized) but I would say that it's something of FreeCAD itself not working properly with `spaces` as separator... it that config common for you? I'm used to Spanish one or american that use the contrary "." and "," but not blank ones...
[image]

BTW, if you use any of them FreeCAD behaves correctly.

Grubuntu, 2024-02-10:
> Great, not it works perfectly.

Did you say 'not' or 'now' ? 

> I confirm that behavior also in normal macOS (I mean not virtualized) but I would say that it's something of FreeCAD itself not working properly with `spaces` as separator... it that config common for you? I'm used to Spanish one or american that use the contrary "." and "," but not blank ones... 

> BTW, if you use any of them FreeCAD behaves correctly.

It working properly with 'spaces' separator (only the "blank" separator does not work)

pgilfernandez, 2024-02-10:
I'm sorry, I meant "now it works perfectly"
=)

Grubuntu, 2024-02-10:
@pgilfernandez @hasecilu 
Thank you very much for your work and testing.

If this looks good to you, I'll push the new version 1.3.8 in a day or two.

closed 2024-02-12 by Grubuntu