Grubuntu/PieMenu #50: [Feature request] Add "ThemeAccentColor" support

pgilfernandez, 2024-02-20:
Latest FreeCAD stylesheets support what they call "ThemeAccentColor", it a pretty nice feature that allow users customize the stylesheets by choosing "accent" colors in FreeCAD Preferences ( Preferences > Display > Theme ).

It would be nice that our styles could get those variables, that is, reading the 3 available accents from Preferences and replace the active stylesheet. Our stylesheets would include (as regular FreeCAD ones do) the variable names `ThemeAccentColor1`, `ThemeAccentColor2` and `ThemeAccentColor3`.

I would say it's not a difficult feature to include and it would be nice for matching our styles with the "accent" colors the user is using inside FreeCAD.

What do you think?

pgilfernandez, 2024-02-20:
I had an inspiration and I'm working on it... I think I will be able to develop it  ;)

Grubuntu, 2024-02-20:
Yes, good idea. Go

pgilfernandez, 2024-02-20:
closed as completed https://github.com/Grubuntu/PieMenu/pull/51

closed 2024-02-20 by pgilfernandez