Grubuntu/PieMenu #54: Pies can be triggered everywhere

pgilfernandez, 2024-02-20:
Pies can be triggered everywhere, that enter in conflict with anywhere you want to type some text and need to use a letter assigned to a pie.

If you are editing a value in the Properties View, editing a macro in the Text Editor, typing some commands in the Python console... when you type a letter the pie opens...

Pies should be restricted to 3D View and maybe somewhere else, I haven't fully think about it but, they can't be triggered when you need your full keyboard key to freely type some text or code.

Grubuntu, 2024-02-21:
I tried to reproduce the problem but it doesn't happen for me.
PieMenus do not open when I edit a property, spreadsheet or macro. I'll do more testing when I have some time.

Grubuntu, 2024-02-21:
I think I fixed the problem: https://github.com/Grubuntu/PieMenu/commit/42b975d271bbe29985b2246bd7e1db23f213d58a

Keep me informed. THANKS

pgilfernandez, 2024-02-21:
now it works as expected, at least under macOS

Thanks!!

closed 2024-02-21 by Grubuntu