Grubuntu/PieMenu #57: Changes are not saved when you open a new instance of FreeCAD

hasecilu, 2024-03-07:
I don't know if this is a bug on PieMenu or it's a consequence of Parameters implementation on FreeCAD.

Lately I have been hacking on external WBs which makes to restart FreeCAD frequently, to load new changes, the point is that I have noticed that when creating a new WB or updating one the changes are intermediately working in the main FreeCAD instance but when the new FreeCAD instance is opened the PieMenu doesn't pick the recent changes.

In the other hand if the firstly opened FreeCAD isntance is closed before the new instance is open the changes prevail.

So, there is in PieMenu some kind of event on closing or is it because some Parameters stuff?

Grubuntu, 2024-03-08:
I noticed that FreeCAD loads the configuration file in memory and modifies it in memory when a parameter is changed. And it's only when FreeCAD is closed (or perhaps at regular intervals) that the configuration file is finally written to the hard disk.

Try this test: Open FreeCAD, manually modify the user.cfg file on the disk and save the file. Then close FreeCAD, and you will see that FreeCAD has overwritten your changes on disk.

FreeCAD works with configuration load in memory, and when PieMenu loads or saves a parameter it also does so in memory, so if you open another instance of FreeCAD, this new instance will read the configuration file from the disk and so this configuration may be different from the configuration currently in memory in the other instance of FreeCAD.

I don't know if this is normal behaviour, or if there is a command to force the parameters to be saved in the file.

hasecilu, 2024-03-08:
It seems that's the case, even the Parameters dialog have a `Save to disk` button, could be that action be done via Python WB?
https://wiki.freecad.org/Std_DlgParameter
`../src/Gui/DlgParamterImp.cpp`
```c++
void DlgParameterImp::onButtonSaveToDiskClicked()
{
    int index = ui->parameterSet->currentIndex();
    ParameterManager* parmgr = App::GetApplication().GetParameterSet(ui->parameterSet->itemData(index).toByteArray());
    if (!parmgr)
        return;

    parmgr->SaveDocument();
}
```

Grubuntu, 2024-03-10:
there's an old thread on the forum:
https://forum.freecad.org/viewtopic.php?style=1&t=45365

You can save the parameters with this command in the python console: 
`App.saveParameter()`

hasecilu, 2024-03-12:
Do you think some folks may find useful for testing changes that behavior or would be better when making changes to use `App.saveParameter()`?

Grubuntu, 2024-03-13:
I think I should be able to integrate direct saving of settings to disk when you change them in PieMenu.

The only negative point is that it increases the disk write rate, perhaps that's why the FreeCAD developers made this choice?

Grubuntu, 2024-03-14:
I've finished integrating the systematic saving of settings to disk. 
Now, as soon as the user closes the preferences window, the settings are saved to disk.

https://github.com/Grubuntu/PieMenu/commit/c89939d4f7a9f0c03ec7474d3bea324e8d9f7723

closed 2024-03-14 by hasecilu