Grubuntu/PieMenu #58: [Discussion] Implementing GUI translation on Crowdin

hasecilu, 2024-03-09:
From what I have seen on the main FreeCAD Crowdin project almost every day FreeCAD is being translated, on the addons project only 7 external WB are there from more than 100!.
The point is that having the same people that translates main FreeCAD translating the addons will help end users by having a level of consistency on the whole FreeCAD environment.

I'm sure doing this fellow translators that are already helping FreeCAD ecosystem would notice the translation support more easily on Crowdin that here on GitHub.

@kaktusus can help us to maintain translations there.
Read this: https://github.com/shaise/FreeCAD_FastenersWB/issues/340

---

Also related to translation, the menu that PieMenu create is called "Accessories" but I think it's better to call it "PieMenu"

[image]

kaktusus, 2024-03-09:
Of course.
I will provide a soft launch. :wink:

Grubuntu, 2024-03-10:
> Also related to translation, the menu that PieMenu create is called "Accessories" but I think it's better to call it "PieMenu"

Thanks,  I'll take a look at it and put it into action.

> Also related to translation, the menu that PieMenu create is called "Accessories" but I think it's better to call it "PieMenu"

This is not a choice linked to PieMenu, other FreeCAD accessories can also use this Menu.
[image]

kaktusus, 2024-03-10:
> Thanks, I'll take a look at it and put it into action.

If you need support in this area, or want to ask something don't hesitate. I'm here,
Greetings

Grubuntu, 2024-03-10:
@kaktusus 
Thanks, I've just asked Yorik to integrate PieMenu on Crowdin. I think that's how I should start?

kaktusus, 2024-03-10:
This is one of the possible ways.
As you know Yorik is swamped with work.
We can do it ourselves by relieving the burden on Yorik.

If you would like to have access to the management of translations to not be cut off from this opportunity create an account on Crowdin and give me your nickname.
This is not mandatory at this time.
If you wish I can introduce you to the procedure. 
Basically, it's two clicks and a heap of fun.

This is how it looks from my point of view. :wink:

hasecilu, 2024-03-10:
> This is not a choice linked to PieMenu, other FreeCAD accessories can also use this Menu. !

OK, got it! Ideally the menu could be also translated but I think if only one WB/tool translate it 2 menus will be created instead of 1. What other tools use Accessories?

Grubuntu, 2024-03-11:
@kaktusus 
> If you would like to have access to the management of translations to not be cut off from this opportunity create an account on Crowdin and give me your nickname. 

Thanks for your help. Here is my Nickname on Crowdin : Grubuntu

kaktusus, 2024-03-11:
I prepared a source file to upload to Crowdin. [PieMenu.zip](https://github.com/Grubuntu/PieMenu/files/14557649/PieMenu.zip) To upload it to Crowdin it must be unpacked. :wink: 
Would you like to do it the first time or should I start the procedure?

The header should look as follows.
```
<?xml version="1.0" encoding="utf-8"?>
<!DOCTYPE TS>
<TS version="2.1">
```

It is important that it does not contain the tags `language` and / or `sourcelanguage`.

```

<?xml version="1.0" encoding="utf-8"?>
<!DOCTYPE TS>
<TS version="2.1" language="pl" sourcelanguage="en">
```

Grubuntu, 2024-03-11:
I'll let you do it. As I have no idea where I should send the file in Crowdin. Thanks

kaktusus, 2024-03-11:
I would lead you by the hand.

In any case, it happened. Welcome PieMenu Addon to the FreeCAD-addons repository on the Crowdin platform. :smile: 
I prepared a presentation maybe it will be useful for the next developers.

[upload and configure new project file](https://github.com/Grubuntu/PieMenu/assets/39754351/d3853138-2383-4760-ba3f-1f0bb14df85c)

Grubuntu, 2024-03-11:
Great, thanks. 
As I'm not a developer on Crowdin, I don't yet have access to all these tabs.
[image]

kaktusus, 2024-03-11:
You are handling Crowdin perfectly. Bravo 
:clap::clap:
:trophy:

closed 2024-03-14 by Grubuntu