Grubuntu/PieMenu #59: Implementing separator in PieMenus

Grubuntu, 2024-03-12:
As requested in this thread :
https://forum.freecad.org/viewtopic.php?t=84101&start=100#p745928

And as I'd already tried to work on it, and I've finally managed to implement the addition of separators in the menus :

[image]

[image]

https://github.com/Grubuntu/PieMenu/commit/f222c33297d9c7627d2ea05b0018b617782277cc

I've chosen to make the selectors and associated text (display command name) completely transparent in all styles, but this can be changed.

Can you try it and tell me what you think?

The PieMenuAddSeparator.svg icon could be improved, what do you think @pgilfernandez ?
[image]

Thank you.

Grubuntu, 2024-03-13:
I found out why it worked for me during my tests: while developing the code, I had created a menu bar that contained the fake 'Separator' command and it worked.

In FreeCAD, if a custom command isn't assigned to a menu bar, PieMenu can't find it...

I'm looking into how to resolve this as best I can.

[image]

pgilfernandez, 2024-03-13:
Umm, I can't even do it manually, "PieMenu" category is not available...

Grubuntu, 2024-03-13:
I've tried to correct the problem, so you can try again. Thanks for your time.
https://github.com/Grubuntu/PieMenu/commit/1bb631ecee469756b377cf0c1fb3f68b5dd1780b

hasecilu, 2024-03-13:
Same here, very neat feature!

closed 2024-03-16 by Grubuntu