Grubuntu/PieMenu #63: Creation of Wiki page

hasecilu, 2024-03-16:


Grubuntu, 2024-03-16:
A great idea.

I'll add a button to the documentation and someone can create the wiki page, or I'll do it.
Thanks

hasecilu, 2024-03-16:
I was thinking on adding:
- Introduction with a picture of a PieMenu
- Installation
- Usage, with the main tabs
- At the end optional: references and links

hasecilu, 2024-03-16:
This is the initial skeleton, if you have Wiki access you can finish it else you can give the media you prefer to use.
https://wiki.freecad.org/PieMenu_Workbench

Grubuntu, 2024-03-17:
Thank you very much for your work

I don't have the rights to edit the wiki at the moment (I'll make the request). Feel free to take the initiative of inserting PieMenu screenshots into the wiki that you think are suitable, if you like. PieMenu is a team effort, so you can add your own personal touch.

hasecilu, 2024-03-27:
I made some progress but there is still missing the context example and the advanced settings.

dzid26, 2024-03-27:
> I made some progress but there is still missing the context example and the advanced settings.

What does Pie menu has to do with workbenches? I don't see the Pie Workbench. 
[image]
Maybe it is possible to remove workbench references from the wiki page?

hasecilu, 2024-03-27:
Maybe, on the Addon Manager it falls under the Workbench group, maybe we need another group like "accesories"

[image]

closed 2024-06-04 by Grubuntu

hasecilu, 2024-11-01:
Can you open the documentation on FreeCAD?
On FreeCAD weekly with Qt5 I got: 

> Could not find QtWebEngineProcess
> /tmp/.mount_freecaltRcxz/AppRun: line 41: 246119 Aborted                 (core dumped) "${MAIN}" "$@"

On FreeCAD with Qt6 PieMenu crashes FreeCAD but of course the documentation can be open in the built-in browser.

Test with:

```python
import Help
>>> Help.show("PieMenu Workbench")
```

Grubuntu, 2024-11-01:
> Can you open the documentation on FreeCAD? 

It works for me on  39057-conda.

> I think Wiki needs and update prior 1.0 release, I'll update pictures. @Grubuntu let me know if you have pieces of information that want added. Personally I never understand the "context" stuff so I don't know what to put there.

Perhaps an example will speak for itself:

In normal use, the default PieMenu opens when the global shortcut is pressed, but sometimes the user needs specific tools depending on the type of 3D object selected. 

e.g : For example, the user may want only the tools that are useful when a face of a 3D object is selected in PartDesign: 

He can create a PieMenu called 'Faces' containing these tools and set the display conditions for this PieMenu in the Context tab as follows:

[image]

If the user selects only one face (while in the PartDesign workbench) and presses the global shortcut,  the ‘Faces’ PieMenu is displayed, giving access only to the tools that the user has set in the ‘Faces’ PieMenu.

[image]

Ticking the box 'Immediate triggering when conditions are met' change the behavior will trigger ticking the box will immediately display the PieMenu without having to press the global shortcut.

hasecilu, 2024-11-18:
Hey, thanks for the tips, I have added that information to the Wiki page, let me know if there is something that needs to be updated.