Grubuntu/PieMenu #66: v1.5 New features to test

Grubuntu, 2024-03-20:
Hi
@pgilfernandez @hasecilu 

I've published a draft of version 1.5 with a few new features:
https://github.com/Grubuntu/PieMenu/tree/next-release150

- Added support for 'Del' and 'Suppr' keys for deleting Tool in Toollist (Preferences):
press 'Del' and 'Suppr' when a tool is selected in toolist, remove it (as the button "Remove" do

- Added validation with 'Enter' and 'Return' keys, when buttons "Validate" and "Cancel" are visible.
(as a side-effect, pressing the Enter and return keys also hides the PieMenu)
[image]

- Added documentation link (documentation under construction, I don't have the editing rights yet, all help is welcome) https://wiki.freecad.org/PieMenu_Workbench
Is 'Documentation' better than 'Help' ?
[image]

- Added expressions binding in fast spinbox 
[image]

- Added toggle show/hide PieMenu behavior also for individuals shortcuts: need to change the title slightly
[image]

- Added tools shortcuts keys (for each shape !):
[image]

[image]
[image]
[image]
[image]

The shortcuts keys are assigned chronologically, starting with the number 0 and incrementing according to the order of the characters in the ASCII table  '0 1 2 3 4 5 6 7 8 9 : ; < = > ? @ A B C etc...' https://www.ascii-code.com/en
It is not possible to customise the shortcuts.

Usage : Simply enable the functionnaly, and press the dedicated shortcut when a PieMenu is open.

- Updating styles according to the new feature 'Tools shortcut': QLabel#pieMenuShortcut

Try, test and tell me what you think. Thanks.
I hope you will enjoy them.

hasecilu, 2024-03-21:
Hi, I have tested the new changes and find this:
- `Del` & `Supr` work, would be useful to have something like `Ctrl + Up/Down` to move the selected command? if so, what about to add a separator?
- The behavior with `Enter` and `Return` seems fine to me
- I'll try to make advancement on Wiki today, sent a new PR also
- Super useful to have expression in the spinbox but I noticed that pressing `Enter`/`Return` no longer closes the task panel.
- Toggle works on main PM and when you are in (let's say) PartDesign and the PartDesign menu opens with the default shortcut but in you want the PartDesign in another WB with its custom shortcut it doesn't toggle
- pie-key shortcuts seems to work fine, Does PieMenu have a way to detect if a shortcut conflict with default shortcuts? Personally I use `<` key but is there is someone using letters could be problems or not?

Grubuntu, 2024-03-21:
Thanks for testing.

> * `Del` & `Supr` work, would be useful to have something like `Ctrl + Up/Down` to move the selected command? if so, what about to add a separator?

Yes, I'm going to think about the possibility of moving the commands around on the keyboard: I'm adding that to my list of things to do.

> * I'll try to make advancement on Wiki today, sent a new PR also

Super !

> * Super useful to have expression in the spinbox but I noticed that pressing `Enter`/`Return` no longer closes the task panel.
 
I'm going to look into it, thanks for pointing it out to me.

> * Toggle works on main PM and when you are in (let's say) PartDesign and the PartDesign menu opens with the default shortcut but in you want the PartDesign in another WB with its custom shortcut it doesn't toggle

Can you explain how to reproduce the problem, I don't see it.

> * pie-key shortcuts seems to work fine, Does PieMenu have a way to detect if a shortcut conflict with default shortcuts? Personally I use `<` key but is there is someone using letters could be problems or not?

No, there is no detection of duplicate shortcuts at this level. I'd thought about limiting the number of shortcuts to just 8 like in Blender... I'll have to do some more tests because I have the impression that some alphabetical shortcuts don't work properly. 
I might have to limit the number of shortcuts to 10 (from 0 to 9).

 I'm still working on it, so please let me know if there are any problems. 

THANKS
Grub

hasecilu, 2024-03-21:
> Can you explain how to reproduce the problem, I don't see it.

1. Check shortcut toggle 
2. Pressing `<` toggles the default PieMenu
3. Pressing `P,D` activates the PartDesign PieMenu but can't deactivate it. Also note that PD is assigned to be the default on PD WB, so pressing `<` toggles the same as with the default PieMenu.
4. Pressing `S, K` activates the Sketcher PieMenu but can't deactivate it.

Is this the expected behavior?

Grubuntu, 2024-03-21:
OK, now I understand : Using multiple-key shortcuts causes problems, it's a known limitation for the moment :
https://github.com/Grubuntu/PieMenu/issues/8

it was a bad idea to implement them, only one-key shortcuts or shortcuts with a modifier (CTRL SHIFT ALT) should be used.

Grubuntu, 2024-03-25:
Thanks to your advice hasecilu, I've corrected a few problems and adjusted a few elements.
I have updated the code:

- Cleaning some redondant code, redefine/factorize some functions (get and set parameter)
- Reorganisation of the code (grouping class, definitions...)
- Added UP and DOWN key for delete and move Tools il toollist (Preferences)
- Added Icon PieMenu to Preference dialog window 
- Fix tools shortcuts keys feature :  shortcut from 0 to 9 and A to Z
- Update stylesheets
- Limits the maximum number of tools in PieMenu to 30 to avoid problems with tools shortcuts (34 or more shortcuts causes unstable behaviour, which I can't explain, so I've deliberately limited to 30 shortcuts)
- Added label and manage toollist (greyed) if maximum number of tool is reached

[image]

[image]

pgilfernandez, 2024-03-27:
Very nice improvements!
I'm not having much time to test or code but they look very cool to me.
I would add the option to hide command shortcuts as once users know the ones that are assigned they may want to hide them while continue using the shortcuts...

hasecilu, 2024-03-27:
> Cleaning some redondant code, redefine/factorize some functions (get and set parameter)

Do you think that the code would be easier to read/navigate if instead of creating the GUI of the PieMenu settings via code the GUI is created with a `.ui` file with Qt Designer? Translations would not be lost. Also what about moving some classes to other files, is possible? Have you considered to use `pre-commit` to have a formatting hook, this is the one used on main FreeCAD https://github.com/FreeCAD/FreeCAD/blob/main/.pre-commit-config.yaml

> Added UP and DOWN key for delete and move Tools in toollist (Preferences)

Working fine.

> * Added Icon PieMenu to Preference dialog window
> 
> * Fix tools shortcuts keys feature :  shortcut from 0 to 9 and A to T (30 shortcuts)
> 
> * Update stylesheets
> 
> * Limits the maximum number of tools in PieMenu to 30 to avoid problems with tools shortcuts (34 or more shortcuts causes unstable behaviour, which I can't explain, so I've deliberately limited to 30 shortcuts)
> 
> * Added label and manage toollist (greyed) if maximum number of tool is reached

Cool

I made #69

Grubuntu, 2024-03-28:
> * Do you think that the code would be easier to read/navigate if instead of creating the GUI of the PieMenu settings via code the GUI is created with a `.ui` file with Qt Designer? Translations would not be lost, in case of any change CrowdIn helps yo "remember".

That's probably true, I've never used QtDesigner.

> * Also what about moving some classes to other files, is possible?

Yes, there are ways of improving the readability and maintenance of the code, and I'm taking it one step at a time.

> * Have you considered to use `pre-commit` to have a formatting hook, this is the one used on main FreeCAD https://github.com/FreeCAD/FreeCAD/blob/main/.pre-commit-config.yaml

I have no idea what it is or how to set it up. I'll read the documentation, thanks.

> I did #69

Thanks !

Grubuntu, 2024-03-30:
>  I would add the option to hide command shortcuts as once users know the ones that are assigned they may want to hide them while continue using the shortcuts...

I added this function in the last release-150 update, thank you for suggesting this idea.

https://github.com/Grubuntu/PieMenu/commit/2f66dac619ce85ee148735b8d01ebe66cc3ed2c1#diff-bde23e3d8118c05e6614f9923e700773a78f6fc9b0fe18ded5fd27cdbec5db7f

I've also modified the Preferences interface so that it follows the FreeCAD graphic codes and a few other adjustments.

[image]
[image]
[image]
[image]

- Matches the style of the Preferences window with the style of FreeCAD Preferences: Groupboxes, Checkboxes left aligned, Spinbox stretched etc
- Added buttons to clear shortcut and globalshortcut
- Highlighting the default PieMenubox by an arrow in front in PieMenu's list and in QuickMenu
- Added checkbox to set the defaut PieMenu in Preferences
- Enable the tools shortcuts will add a column with shortcuts in toollist 
- Added show/hide tools's shortcuts : + spinbox for selected font size
- Removed label and limitation for number of shortcuts in a PieMenu : tools's shortcuts can have all number and letter 0-9 A-Z (except 'X' which bug), beyond 'Z' no shortcuts are allocated.

I'm thinking of freezing the development of this version now.

Feel free to give it a try. Thanks

hasecilu, 2024-03-30:
Looks good, new changes improve UX, specially marking default menu on dialog.

I'll try to recreate the UI on QtDesigner.

closed 2024-04-08 by Grubuntu