Grubuntu/PieMenu #70: Feature - Long right click to open the Pie

dzid26, 2024-03-27:
Long right click to open the Pie -(instead of keyboard shortcut) - I think this is how it works in Fusion 360 and it makes sense. 
Is it possible?

Grubuntu, 2024-03-28:
Hi

When I first developed a macro to display commands quickly, before developing PieMenu, I tried to implement the right-click, but unfortunately Freecad already uses the right-click:
right-click in the sketcher = cancels the current command or displays the context menu
right-click in the 3D view displays a context menu.

This would conflict with the use of the right-click in PieMenu but I'm still thinking about it, I'll give it another try.

hasecilu, 2024-03-28:
Some mouses have 2 extra buttons on the thumb place, maybe trying to use those button is useful.

dzid26, 2024-03-28:
@Grubuntu  Default Freecad actions are performed on the button release.  If it was possible to cancel the default action if the button was not released within a second or so, then the Pie could be opened.  That was my thinking. 

@hasecilu  Oh, I have extra buttons on my mouse. I may use your suggestion :)

Grubuntu, 2024-06-03:
This experimental feature is now include in version 1.6
Because of the way Freecad uses the right click, this can create unexpected mouse behaviour.

closed 2024-06-04 by Grubuntu