Grubuntu/PieMenu #74: Unable to import shiboken2

sdifru, 2024-04-07:
I got this error `PySide2/__init__.py: Unable to import shiboken2 from <<followed by many directories to workbenches>>` after I installed this add-on.

I also couldn't get it to work.

Here is my freecad version:
[code]
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Python 3.11.8, Qt 6.6.3, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Assembly3 0.12.2
  * fasteners 0.5.16
[/code]

Grubuntu, 2024-04-08:
Hello

Sorry I can't test on Arch Linux.

Perhaps it's a Pyside2 dependency problem? https://forum.freecad.org/viewtopic.php?t=27805

hasecilu, 2024-04-08:
Do you have `extra/python-shiboken2` package installed?

sdifru, 2024-04-11:
I have both `pyside2` and `python-shiboken2` installed @Grubuntu @hasecilu. Could it be that it doesn't load dependencies from system installed packages?

hasecilu, 2024-04-11:
Try running on FreeCAD console
```python
import sys
print('\n'.join(sys.path))
```
It shows the places from where it read the libraries, including on WB dirs. In my case it seems all Qt, Pyside packages are in `/usr/lib/python3.11/`, it appears on FreeCAD and it works.

sdifru, 2024-04-12:
It prints some directories and here are some of them:
```
/usr/lib/python3.11
/usr/lib/python3.11/lib-dynload
/usr/lib/python3.11/site-packages
```
Using `fd shiboken2`, I found these:
```
/usr/lib/python3.11/site-packages/shiboken2/
/usr/lib/python3.11/site-packages/shiboken2.egg-info/
/usr/lib/python3.11/site-packages/shiboken2_generator/
```
Using `fd pyside2`, I found these:
```
/usr/lib/python3.11/site-packages/PySide2/
/usr/lib/python3.11/site-packages/PySide2.egg-info/
```

@hasecilu do you use arch linux? If so, do you install freecad by `sudo pacman -S freecad` or other means?

hasecilu, 2024-04-12:
Yes, but lately I just use compiled dev version.

It seems FreeCAD was ported to Qt6 and that's causing some issues, a lot of changes recently, there are some reports on the issues section: https://gitlab.archlinux.org/archlinux/packaging/packages/freecad/-/commit/15263d56dbd4adb0e3e227fd7b534e731c05522b

If that's the cause of the problems maybe trying to use the weekly appimage or compiling would be better, I suppose either flatpak or snap should work.

sdifru, 2024-04-14:
I downgraded the freecad version and it works. It seems like moving to QT6 causes some problems for arch linux freecad users for now.

closed 2024-04-14 by sdifru