Grubuntu/PieMenu #76: Using pyside6

sdifru, 2024-04-30:
It seems like freecad now use qt6 and pyside6. I'm on arch linux and I got this error: `During initialization the error "No module named 'PySide2'" occurred in /home/myname/.local/share/FreeCAD/Mod/PieMenu/./InitGui.py`.

Could PieMenu use pyside6 instead of pyside2?

Grubuntu, 2024-04-30:
It seems that we can now use pyside instead of pyside2, it  will call the pyside version which is in FreeCAD: 
https://forum.freecad.org/viewtopic.php?t=85089#p739422

Try this :  at the beginning of your InitGui.py, (/home/myname/.local/share/FreeCAD/Mod/PieMenu/./InitGui.py) edit the 3 lines:

```
    from PySide2.QtGui import QKeyEvent, QFontMetrics
    from PySide2.QtGui import QKeySequence
    from PySide2.QtCore import Qt
```

replace them by :
```

    from PySide.QtGui import QKeyEvent, QFontMetrics
    from PySide.QtGui import QKeySequence
    from PySide.QtCore import Qt
```

let me know if it works correctly I'll update for future versions.
Thanks

sdifru, 2024-05-03:
Freecad crashed with message: `Program received signal SIGSEGV, Segmentation fault.`

hasecilu, 2024-05-05:
I'm able to use PieMenu on Arch Linux using the latest weekly build AppImage, but `freecad` package (FC 0.21) is totally broken, also can't even compile.

Grubuntu, 2024-05-16:
This seems to be linked to the FreeCAD 0.21 and Qt6 code. We'll have to wait for an update to FreeCAD 0.21.

https://github.com/FreeCAD/FreeCAD/issues/13303
https://github.com/FreeCAD/FreeCAD/issues/13427

Seems to work with 0.22dev

luzpaz, 2024-10-20:
No work will be done upstream to address 0.21.x issues (only 1.0 going forward). This issue therefore should probably be closed.

closed 2024-10-20 by Grubuntu