Grubuntu/PieMenu #81: Black FreeCAD window on PieMenu activation

Atenath, 2024-06-03:
Don't know if it's just me but I can't fully try this wonderfully looking addon. Because everytime I activate it with TAB or custom assigned shortcut I get a black screen: 

[image]

Then next time I activate again, and it shows previous locations when I activated this menu +new one. And it happens everytime.

[image]

Am I the only one having this issue? Using Linux Mint with FreeCAD DEV version. But in 0.21.2 I get same results.

Grubuntu, 2024-06-04:
Hi

First thing you can try, in FreeCAD Setting try different stylesheets.

I will try to reproduce the problem in a Virtual Machine, can you tell me which Window Manager you use?

Grubuntu, 2024-06-04:
I've just install a fresh VM with Linux Mint 21.3 Cinnamon, it works fine for me with FreeCAD 0.22Dev, I can't reproduce the issue...
[image]

Perhaps a driver or hardware issue...
I've seen this before : https://forum.freecad.org/viewtopic.php?t=29108&start=50#p498306

Try to update your system, check your Python version, but without more clues, I don't know where to look.

[image]

Grubuntu, 2024-06-04:
I don't know if this has anything to do with it, but it could be a clue: https://forum.freecad.org/viewtopic.php?t=36087&start=40

Atenath, 2024-06-04:
Well I've tried your suggestions. First what I've tried to do is using Nvidia driver 535 switched to Wayland (Experimental) problem solved. Also i've tried to revert to default and then update nvidia drivers (545) but it failed to install so after restart it changed to default opensource nouveau drivers and also there were no problems.

Then after two lucky shots I've decided to go back to Nvidia 535 and default X11. And PieMenu dark screen appears again. 

I feel a little bit silly because I forget to check those simple things first. But at least now we can be sure it's not a PieMenu bug it has to do something with drivers. And looks like both links you provided (quick readthroug) suggest the same issues.

Grubuntu, 2024-06-05:
Thanks for your reply, I happy to see that you have solved your issue.

closed 2024-06-05 by Grubuntu