Grubuntu/PieMenu #83: In PartDesign and Sketcher workbenches, Pie menu shows error

raindropsfromsky, 2024-06-22:
I have installed Pie Menus in Ondsel weekly.

In Part Design WB, I created a body and then created a sketch. 
So I was in Sketcher WB, but had not added any sketch yet.
I pressed TAB, and got a pie menu.

But at the same time, I also got this error:

QWindowsWindow::setMouseGrabEnabled: Not setting mouse grab for invisible window QWidgetWindow/'styleContainerWindow'

Worse, the message panel shows a gray panel also (duplicate message). 
[image]

I added a sketch and closed the sketch.
So I was in the PD WB at this moment.
I pressed TAB again, and got the same error.

**Desired:** 
1. There should be no error
2. If there is any error, show only it only once (not in two duplicate forms) 

Using Ondsel 37608 (latest weekly)

Grubuntu, 2024-06-22:
Hello,

Thank you for your feedback. 

It's true that this error has been there for some time and I haven't managed to get rid of it yet. I'm going to try and spend as much time as I need to solve the problem.

In the meantime, you can uncheck this box to stop being bothered by the warnings.

[image]

Grubuntu, 2024-06-23:
Don't forget to set up the notification area as you wish:

[image]

closed 2024-06-26 by Grubuntu