Grubuntu/PieMenu #87: 'Submenu' in PieMenu

rawoki, 2024-07-02:
Thank you very much for developing this add-on. This is a very useful program.
I really like using 'submenus' in PieMenu because I can limit the number of icons displayed at one time. This works well in PartDesign, but unfortunately in the Sketcher the 'submenu' is gray and cannot be expanded. Can you make the 'submenu' always active regardless of the workbench?

Grubuntu, 2024-07-02:
Hello, 

Thank you for your message. 

Unfortunately, due to FreeCAD behaviour when you enter in "edit mode" in the Sketcher, some commands are disabled, which also disables the 'Submenus'. 

Some commands are available when editing a sketch and others are not, I'm going to try and find a solution to make the Submenus available in this mode.

I'm going to try and work on this over the next few days. Thank you

Grubuntu, 2024-07-02:
Hi

I've solved the problem and released version 1.7.4, which fixes it. 
You can uninstall and reinstall the PieMenu to update to the latest version.

Thanks.

closed 2024-07-02 by Grubuntu

rawoki, 2024-07-03:
Thanks. It was very fast.  It works perfectly.