Grubuntu/PieMenu #92: [Feature request] Export/Import pies config

pgilfernandez, 2024-07-31:
Hi!

Now that the pues are so great, it feels like a good moment to implement an import/export option so that pies can be shared or backup...

What I'm not sure if it would be a good idea to split the export in the following:
- Single pie
- All pies
- Global settings

What do you think?
Cheers

Grubuntu, 2024-07-31:
Hi !

Yes, that's a very good idea, I'd started thinking about it when I was asked: 
https://forum.freecad.org/viewtopic.php?t=84101&start=120#p770003

Initially, I'm thinking of working on exporting all the Piemenu settings, then I'll look at adding an individual export. 
I'll work on it as soon as I have a bit more time.

Grubuntu, 2024-08-18:
Version 1.8 now include ability to import/export global parameters.

[image]
https://github.com/Grubuntu/PieMenu/pull/93

Thanks for your support.

closed 2024-11-11 by Grubuntu