Grubuntu/PieMenu #98: PieMenu crashes FreeCAD at startup when FreeCAD is build with QT6.

APEbbers, 2024-10-08:
When I use PieMenu with a FreeCAD version that is build with QT6 and Python 3.12, FreeCAD crashes at atartup.
The log shows:
[image]
When I use a FreeCAd build with QT5, there is no problem.

I'm using Windows 11 and the develop version of FreeCAD 1.0.

Kind regards,

Paul Ebbers

Grubuntu, 2024-10-09:
Hi

Where did you find your Qt6 version? I'm not even sure if FreeCAD currently supports Qt6?
I don't have a Qt6 version so I can't test it at the moment.

APEbbers, 2024-10-09:
Hi, I believe the nigthly builds have QT6. or you can compile your own FreeCAD with the latest commits.
I tried to look for the cause meself, but I couldn't find a clear cause.

Just a heads up:tThe new FreeCAD 1.0 wil have QT6 compatibility. 

kind regards Paul

hasecilu, 2024-11-01:
Hi @Syres916, hope to not bother you, do you think you could help us with this in the future, it's a problem that has been occurring for some months already.

I made compilation with Qt6 with debug mode on and got the trace below using `gdb`, does it give any clue? It mentions a file at the end, anyway I don't know much about Qt.

I'm Using Arch Linux, latest libraries/packages, compiling from `HEAD`. Any other info needed?

<details>
  <summary><b>Click here to show/hide flags</b></summary>

cmake -S "$source_dir" -B "$build_dir" \
	-D BUILD_FLAT_MESH=ON \
	-D OCCT_CMAKE_FALLBACK=1 \
	-D FREECAD_QT_VERSION=6 \
	-D CMAKE_BUILD_TYPE=Release \
	-D CMAKE_C_FLAGS="$CFLAGS -ffat-lto-objects -fPIC -w" \
	-D CMAKE_CXX_FLAGS="$CXXFLAGS -ffat-lto-objects -fPIC -w" \
	-D FREECAD_USE_EXTERNAL_PIVY=ON \
	-D FREECAD_USE_QT_FILEDIALOG=OFF \
	-D FREECAD_BUILD_DOXYGEN=ON \
	-D CMAKE_BUILD_TYPE=Debug \
	-G Ninja
</details>

```gdb
$ gdb /opt/FreeCAD_1_0_Qt6/bin/FreeCAD
GNU gdb (GDB) 15.2
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/FreeCAD_1_0_Qt6/bin/FreeCAD...
(gdb) run
Starting program: /opt/FreeCAD_1_0_Qt6/bin/FreeCAD

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.archlinux.org>
Enable debuginfod for this session? (y or [n]) y
Debuginfod has been enabled.
To make this setting permanent, add 'set debuginfod enabled on' to .gdbinit.
Downloading separate debug info for /lib64/ld-linux-x86-64.so.2
Downloading 13.47 M separate debug info for /usr/lib/libhdf5.so.310
Downloading 92.82 K separate debug info for /usr/lib/libsz.so.2
Downloading 194.53 K separate debug info for /usr/lib/libz.so.1
Downloading 25.68 M separate debug info for /usr/lib/libmpi.so.40
Downloading 16.40 M separate debug info for /usr/lib/libxerces-c-3.2.so
Downloading 2.52 M separate debug info for /usr/lib/libm.so.6
Downloading 23.62 M separate debug info for /usr/lib/libpython3.12.so.1.0
Downloading 1.17 M separate debug info for /usr/lib/libfmt.so.11
Downloading 2.08 M separate debug info for /usr/lib/libQt6Xml.so.6
Downloading 45.91 M separate debug info for /usr/lib/libCoin.so.80
Downloading 1.89 M separate debug info for /usr/lib/libboost_filesystem.so.1.86.0
Downloading 3.60 M separate debug info for /usr/lib/libboost_program_options.so.1.86.0
Downloading 2.66 M separate debug info for /usr/lib/libboost_regex.so.1.86.0
Downloading 7.54 K separate debug info for /usr/lib/libboost_system.so.1.86.0
Downloading 1.33 M separate debug info for /usr/lib/libboost_thread.so.1.86.0
Downloading 8.81 K separate debug info for /usr/lib/libboost_date_time.so.1.86.0
Downloading 266.10 K separate debug info for /usr/lib/libGL.so.1
Downloading 12.56 M separate debug info for /usr/lib/libQt6PrintSupport.so.6
Downloading 2.20 M separate debug info for /usr/lib/libQt6SvgWidgets.so.6
Downloading 9.89 M separate debug info for /usr/lib/libQt6Svg.so.6
[... 226 lines cut ...]
[New Thread 0x7fffb3e006c0 (LWP 588655)]
[New Thread 0x7fffb34006c0 (LWP 588656)]
[New Thread 0x7fffb2a006c0 (LWP 588657)]
[New Thread 0x7fffb20006c0 (LWP 588658)]
[New Thread 0x7fffb16006c0 (LWP 588659)]
[New Thread 0x7fffb0c006c0 (LWP 588660)]
[Thread 0x7fffb0c006c0 (LWP 588660) exited]
[Thread 0x7fffb16006c0 (LWP 588659) exited]
[Thread 0x7fffb20006c0 (LWP 588658) exited]
[Thread 0x7fffb2a006c0 (LWP 588657) exited]
[Thread 0x7fffb34006c0 (LWP 588656) exited]
[Thread 0x7fffb3e006c0 (LWP 588655) exited]
Downloading 236.63 K separate debug info for /usr/lib/python3.12/lib-dynload/_json.cpython-312-x86_64-linux-gnu.so
Downloading 310.59 K separate debug info for /usr/lib/python3.12/lib-dynload/_elementtree.cpython-312-x86_64-linux-gnu.so
Downloading 719.66 K separate debug info for /usr/lib/python3.12/lib-dynload/_ctypes.cpython-312-x86_64-linux-gnu.so
NOTICE: Addon "/home/uli/.local/share/FreeCAD/Mod/ThreadProfile" disabled by presence of ADDON_DISABLED stopfile
Downloading 21.91 M separate debug info for /usr/lib/python3.12/site-packages/pivy/_coin.so

Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
Downloading 37.73 K source file /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qboxlayout.cpp
QBoxLayoutPrivate::setupGeom (this=this@entry=0x555558dc12d0) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qboxlayout.cpp:253
253             QSize max = box->item->maximumSize();
(gdb)

```

Grubuntu, 2024-11-03:
I had a lot of trouble compiling FreeCAD with Qt6 for Windows, but finally I got a version that works more or less well for testing.

I finally figured out what was wrong with PieMenu:
 in Qt5 the tab numbering can start at 1 whereas with Qt6 it crashes if it doesn't start at 0... so I patched the current version of PieMenu 1.9.1 to avoid a crash when FreeCAD Qt6 starts. 

https://github.com/Grubuntu/PieMenu/commit/326942dc8dd30401916355343f68223c5cdc7c98

However, there are now a lot of other problems with Qt6:

```
self.menu.setWindowFlags(QtCore.Qt.FramelessWindowHint | Qt.NoDropShadowWindowHint | QtCore.Qt.WA_MacAlwaysShowToolWindow)
During initialization the error "unsupported operand type(s) for |: 'WindowType' and 'WidgetAttribute'
```

```
PieMenuInstance.add_commands(actions, context, text)
22:09:11                                   ^^^^^^^
22:09:11  UnboundLocalError: cannot access local variable 'actions' where it is not associated with a value
```
```

22:09:58      modifier_text = self.get_modifier_text(event.modifiers())
22:09:58                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
22:09:58    File "C:\Users\Victus\AppData\Roaming\FreeCAD\Mod\PieMenu\.\InitGui.py", line 246, in get_modifier_text
22:09:58      for modifier in modifier_names if modifiers & modifier])
22:09:58                                        ~~~~~~~~~~^~~~~~~~~~
22:09:58  TypeError: unsupported operand type(s) for &: 'KeyboardModifier' and 'Key'
```

 I'll be working on them over the next few weeks.

Thanks

closed 2024-11-03 by Grubuntu

hasecilu, 2024-11-03:
Cool, another error means progress. Certainly still not totally fixed but no longer crashes, thanks.