Grubuntu/PieMenu #102: [Feature request] Default size settings

chiboreache, 2024-11-06:
making changes in size and radius for every pie is _a bit tedious_

Grubuntu, 2024-11-08:
Hi

Independent customisation of each menu is an important feature of PieMenu.

Can you explain what you would like?

dgrabulosa, 2024-11-09:
You can duplicate a pie menu and edit the tools.

chiboreache, 2024-11-10:
> Independent customisation of each menu is an important feature of PieMenu.

i can't image that someone would ever change that size of pie's independently for every menu `~_~`

> Can you explain what you would like?

Blender is perfect example, you have one setting for all, and you can overwrite if needed

[image]

[image]

[image]

Grubuntu, 2024-11-11:
OK, I see.

Personally, I use different shapes and therefore different sizes of menus, which allows me to differentiate between the basic menus (design), the contextual menus (when a Face or Edge or Vertex is selected), the workbench access menus etc...

[image]

I don't know if it's commonly used because I've had very little feedback from users.

closed 2024-12-24 by Grubuntu