Grubuntu/PieMenu #106: FreCad crash pie menu related.

dgrabulosa, 2024-11-09:
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Spanish/Spain (es_ES)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FreeCAD-Ribbon 1.3.17.1
  * PieMenu 1.9.2
  * sheetmetal 0.5.3

[image]

Grubuntu, 2024-11-10:
thank you for your report.

I will work on this.

Grubuntu, 2024-11-10:
@dgrabulosa 
I can't reproduce the first error at this moment : can you explain how you get it, please?

I can reproduce the second error: In sketcher when 'Sketcher_Dimension' tool is active and if i close the sketcher with green tick, it creates a violation error. Do you only get this error with the Sketcher_Dimension' tool or in other circumstances?

Grubuntu, 2024-11-10:
First error is normaly fix with v1.9.3

I'm still working on the second problem.

Grubuntu, 2024-11-11:
Fixed with v1.9.4

commit e0536db9 (2024-11-10): Merge pull request #107 from Grubuntu/Grubuntu-patch-193

Update to v1.9.3
Fix #104
Partialy fix #106
closed 2024-11-11 by Grubuntu