Grubuntu/PieMenu #112: [Feature Request] Change labels for options in PieMenu

solajim, 2024-11-21:
If possible, would you be willing to let us choose labels for each shortcut?

I've included a screenshot below to explain:
[image]

Grubuntu, 2024-11-21:
Yes, In the next version, I'm going to remove the ‘PieMenu’ header, which will no longer be necessary as I've included a possible sort by WB :

[image]

Grubuntu, 2024-11-22:
‘PieMenu_’ header has been removed from nested PieMenus in V1.10
[image]