Grubuntu/PieMenu #114: Using `What's this?` function crashes FreeCAD when PieMenu is installed

hasecilu, 2024-11-23:
when the option to open the Wiki page is either in a FreeCAD tab or a window.

[image]

Very early PieMenu versions didn't crashed FreeCAD when using `What's this?`, but then around 1.5 version FreeCAD crashed instantly on Qt6 builds, recently changes make Pie Menu usable again on Qt6 builds but now this new crash appear.

I can not properly do a `git bisect` because of the older crash. Also can not test on Qt5 because Arch Linux packages Qt6 version and AppImages with Qt5 says: "Could not find QtWebEngineProcess" so I have no way to know which commit introduced the problem.

I don't find any obvious relation that is causing the problem. I already reset my FreeCAD user config file and the crash only appears when PieMenu is added (I'm disabling it with `ADDON_DISABLED` file). I removed the part where we use the `Help.show()` function but nothing changes.

Apart from this the usage of PieMenu goes really good, the workaround for this is to use your web browser for the Help module.

```
OS: Arch Linux (XFCE/openbox/xcb)
Architecture: x86_64
Version: 1.1.0dev.36460 +2833 (Git)
Build type: Release
Branch: test/prs
Hash: 75283a56a9982657b98c248c33134d78aa141347
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: Spanish/Mexico (es_MX)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * PieMenu 1.10.0
```

### backtrace of SIGSEGV on Qt6 build

<details>
  <summary><b>Click here to show/hide</b></summary>

```gdb
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
0x00007ffff25e2754 in ?? () from /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8
(gdb) bt
#0  0x00007ffff25e2754 in ??? () at /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8
#1  0x00007ffff25e4a28 in PySide::getWrapperForQObject(QObject*, _typeobject*) () at /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8
#2  0x00007fffb301760d in ??? () at /usr/lib/python3.12/site-packages/PySide6/QtWidgets.cpython-312-x86_64-linux-gnu.so
#3  0x00007fffef1553d8 in QCoreApplicationPrivate::sendThroughApplicationEventFilters
    (this=this@entry=0x555555aaf450, receiver=receiver@entry=0x55555ef97c30, event=event@entry=0x7fffffff6e60) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1284
#4  0x00007ffff00fe35a in QApplicationPrivate::notify_helper (this=0x555555aaf450, receiver=0x55555ef97c30, e=0x7fffffff6e60)
    at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3265
#5  0x00007ffff4f95e20 in Gui::GUIApplication::notify (this=0x7fffffffc650, receiver=0x55555ef97c30, event=0x7fffffff6e60) at /home/uli/git/FreeCAD/FreeCAD/src/Gui/GuiApplication.cpp:93
#6  0x00007fffef1585a8 in QCoreApplication::notifyInternal2 (receiver=0x55555ef97c30, event=0x7fffffff6e60) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1165
#7  0x00007fffef1acbb4 in QCoreApplication::sendEvent (receiver=0x55555ef97c30, event=0x7fffffff6e60) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1609
#8  QObject::doSetProperty (this=0x55555ef97c30, name=<optimized out>, lvalue=<optimized out>, rvalue=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qobject.cpp:4307
#9  0x00007ffff25e4acc in PySide::getWrapperForQObject(QObject*, _typeobject*) () at /usr/lib/libpyside6.cpython-312-x86_64-linux-gnu.so.6.8
#10 0x00007fffb301760d in ??? () at /usr/lib/python3.12/site-packages/PySide6/QtWidgets.cpython-312-x86_64-linux-gnu.so
#11 0x00007fffef1553d8 in QCoreApplicationPrivate::sendThroughApplicationEventFilters
    (this=this@entry=0x555555aaf450, receiver=receiver@entry=0x55555ef97c30, event=event@entry=0x7fffffff7440) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1284
#12 0x00007ffff00fe35a in QApplicationPrivate::notify_helper (this=0x555555aaf450, receiver=0x55555ef97c30, e=0x7fffffff7440)
    at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3265
#13 0x00007ffff4f95e20 in Gui::GUIApplication::notify (this=0x7fffffffc650, receiver=0x55555ef97c30, event=0x7fffffff7440) at /home/uli/git/FreeCAD/FreeCAD/src/Gui/GuiApplication.cpp:93
#14 0x00007fffef1585a8 in QCoreApplication::notifyInternal2 (receiver=0x55555ef97c30, event=0x7fffffff7440) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1165
#15 0x00007fffef1585ed in QCoreApplication::sendEvent (receiver=<optimized out>, event=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1609
#16 0x00007fff53b5a9de in QQuickDeliveryAgentPrivate::setFocusInScope (this=<optimized out>, scope=<optimized out>, item=<optimized out>, reason=<optimized out>, options=...)
    at /usr/src/debug/qt6-declarative/qtdeclarative/src/quick/util/qquickdeliveryagent.cpp:508
#17 0x00007fff539a7c1d in QQuickItem::setFocus (this=0x55555efbfeb0, focus=<optimized out>, reason=Qt::OtherFocusReason) at /usr/include/qt6/QtCore/qflags.h:73
#18 0x00007fff53a5823c in QQuickWindow::focusInEvent (this=<optimized out>, ev=<optimized out>) at /usr/src/debug/qt6-declarative/qtdeclarative/src/quick/items/qquickwindow.cpp:239
--Type <RET> for more, q to quit, c to continue without paging--
#19 0x00007ffff01510c9 in QWidget::event (this=0x55555efb6a60, event=0x7fffffff7b10) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidget.cpp:9086
#20 0x00007ffff00fe31a in QApplicationPrivate::notify_helper (this=<optimized out>, receiver=0x55555efb6a60, e=0x7fffffff7b10)
    at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3294
#21 0x00007ffff4f95e20 in Gui::GUIApplication::notify (this=0x7fffffffc650, receiver=0x55555efb6a60, event=0x7fffffff7b10) at /home/uli/git/FreeCAD/FreeCAD/src/Gui/GuiApplication.cpp:93
#22 0x00007fffef1585a8 in QCoreApplication::notifyInternal2 (receiver=0x55555efb6a60, event=0x7fffffff7b10) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1165
#23 0x00007fffef1585ed in QCoreApplication::sendEvent (receiver=<optimized out>, event=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1609
#24 0x00007ffff010b4c8 in QApplicationPrivate::setFocusWidget (focus=0x55555efb6a60, reason=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:1543
#25 0x00007ffff015fc76 in QWidget::setFocus (this=0x55555efb6a60, reason=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidget.cpp:6580
#26 0x00007ffff02d41d3 in QMdiSubWindowPrivate::setActive (this=this@entry=0x55555eef30f0, activate=activate@entry=true, changeFocus=changeFocus@entry=true)
    at /usr/src/debug/qt6-base/qtbase/src/widgets/widgets/qmdisubwindow.cpp:1451
[... 115 lines cut ...]
#108 0x00007ffff016a768 in QWidgetWindow::handleMouseEvent (this=0x55555a4a8430, event=0x7fffffffbc00) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidgetwindow.cpp:632
#109 QWidgetWindow::event (this=0x55555a4a8430, event=0x7fffffffbc00) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidgetwindow.cpp:298
#110 0x00007ffff00fe31a in QApplicationPrivate::notify_helper (this=<optimized out>, receiver=0x55555a4a8430, e=0x7fffffffbc00)
    at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3294
#111 0x00007ffff4f95e20 in Gui::GUIApplication::notify (this=0x7fffffffc650, receiver=0x55555a4a8430, event=0x7fffffffbc00) at /home/uli/git/FreeCAD/FreeCAD/src/Gui/GuiApplication.cpp:93
#112 0x00007fffef1585a8 in QCoreApplication::notifyInternal2 (receiver=0x55555a4a8430, event=0x7fffffffbc00) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1165
#113 0x00007fffef1585fd in QCoreApplication::sendSpontaneousEvent (receiver=<optimized out>, event=<optimized out>)
    at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1623
#114 0x00007fffef785d4b in QGuiApplicationPrivate::processMouseEvent (e=0x55555ead8160) at /usr/src/debug/qt6-base/qtbase/src/gui/kernel/qguiapplication.cpp:2461
#115 0x00007fffef801284 in QWindowSystemInterface::sendWindowSystemEvents (flags=flags@entry=...) at /usr/src/debug/qt6-base/qtbase/src/gui/kernel/qwindowsysteminterface.cpp:1114
#116 0x00007fffe4b8fb87 in xcbSourceDispatch (source=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/plugins/platforms/xcb/qxcbeventdispatcher.cpp:57
#117 0x00007fffeb50d559 in g_main_dispatch (context=0x7fffdc000f30) at ../glib/glib/gmain.c:3357
[... cut ...]
#125 0x00007ffff4df288f in (anonymous namespace)::tryRunEventLoop (mainApp=...) at /home/uli/git/FreeCAD/FreeCAD/src/Gui/Application.cpp:2196
#126 0x00007ffff4df2be9 in (anonymous namespace)::runEventLoop (mainApp=...) at /home/uli/git/FreeCAD/FreeCAD/src/Gui/Application.cpp:2225
#127 0x00007ffff4df2f93 in Gui::Application::runApplication () at /home/uli/git/FreeCAD/FreeCAD/src/Gui/Application.cpp:2299
#128 0x00005555555795eb in main (argc=1, argv=0x7fffffffce68) at /home/uli/git/FreeCAD/FreeCAD/src/Main/MainGui.cpp:330
```
</details>

Grubuntu, 2024-11-23:
related to this issue ?

https://github.com/FreeCAD/FreeCAD/issues/14922

hasecilu, 2024-11-23:
I'm thinking it's an only Linux problem, even only Linux+Qt6.

The FreeCAD issue is when the old Web WB was removed, now we're using `QtWebEngineWidgets`.
```python
>>> from PySide import QtWebEngineWidgets
>>> QtWebEngineWidgets.__file__
'/usr/lib/freecad/Ext/PySide/QtWebEngineWidgets.py'
```

closed 2024-12-24 by Grubuntu