Grubuntu/PieMenu #118: [Feature request]

dgrabulosa, 2024-12-14:
I play with contextual menu and the condition to auto trigger.
I miss the possivility to trigger with conditional statemets in between  conditions something like these mokup .
For example if I select two edges it tiggers a menu, if I select one vertex and one edge triggers the same menu or other diferent.....
[image]

Grubuntu, 2024-12-14:
It's a good idea, I'm going to think about how to implement it in the code. 

Maybe a formula input box, but that would require syntax checking, or the possibility of adding one or more arrays to create multi-conditions...

dgrabulosa, 2024-12-14:
Some test button or something that checks the formula befor apply.

dgrabulosa, 2024-12-19:
Some more things:
when make custom auto trigger pie menu one thing to help determine exact menu trigger are the type of edges, vertex to take in account: normal edges, construction edges, projected edges.

(A crazy idea: parametric or dynamic pie menu: a special menu that the menu tools are configured with conditional statements individually (only one menu configurations that taking in account the selected entities to show the available tools to manipulate these concreted selected entities)  ex: select 2 points only shows the constraint coincident tool with the conditional: points == 2 but not the parallel constraint that have the conditional edges ==2)

Grubuntu, 2024-12-19:
I've started to work on the possibility of using several conditions, but it's going to take a bit of time.

I'd already thought about a contextual menu but it still seems complicated at the moment, unless we have an internal FreeCAD mapping table that lists all the possible commands depending on the selection, but we don't have that option at the moment. I'll keep this idea aside for later.

Grubuntu, 2024-12-21:
V1.11 is now available with multi rules context. 

@dgrabulosa try it and tell me what you think about.

Thank you for this great idea !

closed 2024-12-24 by Grubuntu