Grubuntu/PieMenu #122: Problem on activating Pie Menu with shortcuts

itarozzi, 2025-04-01:
## Shortcut keys open Pie Menu 1 time every 2 key pressed

I installed and configured pie menu. It works, but the Pie Menu is shown only if I press the shortcut twice. The same with global and specific shortcut.

Actually, I think the Pie Menu is activated (the Plasma panel show a new window opened for FreeCAD) but not displayed. 

I added print debug in showAtMouseInstance() function and it is called, with a valid mouse position and seems to execute the for cycle for each button. The print seems the same when the Pie Menu is displayed or not.

The same in FreeCAD 1.0.0 and 1.1.0dev

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.13.2, Qt 6.8.2, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * DynamicData 2.68.0
  * fasteners 0.5.29
  * Nodes 0.1.36
  * lattice2 1.0.0
  * Render 2024.9.25
  * OpenTheme 2024.9.1
  * sheetmetal 0.5.3
  * kicadStepUpMod 11.4.1
  * PieMenu 1.11.2
  * CurvedShapes 1.0.12

```

Grubuntu, 2025-04-01:
Thank you for your report.

Have you assigned the same "global" and "specific" shortcut key ? If yes, try to assign different shortcut.

Grubuntu, 2025-04-01:
I've tried to reproduce the problem in a virtual machine and I manage to reproduce it when the two shortcuts (global and specific) are the same, but only for certain keyboard keys, not all of them.

Could you export your Piemenu configuration (Piemenu settings > Global settings > Backup settings > Export) and post it here?

Thanks

itarozzi, 2025-04-01:
Thank you for your fast reply :)

No, I'm not using the same shortcuts. But I left some shortcuts empty. 

So, I tried to assign shortcuts for each PieMenu, and started to working.  But is not very useful because I prefere to have a global shortcut calling the PieMenu by context.

So I removed again some specific shortcut and still working. Really strange.

May be I done some mistake in previous config.

If you want I exported the configuration before to fix it.

Grubuntu, 2025-04-01:
Thank you for your quick reply.

[image]

Well, double shortcuts (as Q,Q or V,V etc) have always caused problems in PieMenu, I've never managed to handle them properly, it's a known limitation: https://github.com/Grubuntu/PieMenu/issues/8

Try to use single-key shortcuts if possible, or if not, use CTRL, ALT, SHIFT. 

Managing shortcuts is not easy, as Freecad already uses a large number of shortcuts, so it can sometimes have problems with collisions.

itarozzi, 2025-04-01:
Thanks for your comment.
As I wrote, now it seems to works and I'm not able to reproduce .

I think you can close the issue if is hard to know which combination is problematic.

For my next menu I will use single key as you suggest, avoiding to use the same used in global shortcut.

Thank you very much

closed 2025-04-02 by Grubuntu