Grubuntu/PieMenu #126: [Feature Request] Quick Menu Spinbox optional

cjalternativo, 2025-05-03:
I really love this plugin, it greatly speeds up my workflow and offers many advantages for speed modeling. However, I've encountered two issues with the quickmenu spinbox in operations like Pad, Pocket, and Revolution which are the most recurrent operations that I use:

1) The default focus goes to the side panel, so it’s necessary to manually click on the spinbox if you want to edit the dimension. Sometimes clicking accurately can be difficult, and double-clicking the number to edit it doesn’t always work reliably. Additionally, the options within the spinbox are often placed very close together, with very little margin between them, making them hard to read and click in Ubuntu interface (I’ve attached an image to illustrate this). From a user’s perspective, all of this can slow down the modeling process. It would be more convenient if the default focus was set to the spinbox.

2) Another issue is that when using the side panel to adjust additional settings, it often overrides the configurations set in the quickmenu spinbox. This can be a bit frustrating if you’ve already configured some options in the spinbox beforehand. It would be useful if the settings were preserved or shared between both the side panel and the spinbox.

All the other features of the plugin are excellent. However, for these reasons, I've gotten used to working directly in the side panel and avoiding the spinbox. I was wondering if it would be possible to add an option in the general settings to make the spinbox optional, so that users can choose to disable it until it becomes more developed and stable

[image]

Additional details:
- Operating System: Ubuntu 22.04
- FreeCAD 1.0.0
- PieMenu 1.11

Grubuntu, 2025-05-04:
HI 

Thank you for your message, it makes me happy to know that PieMenu is useful for your workflow.

For the focus problem, this is a problem that appears with some Linux distributions (Ubuntu, Mint, Arch)  (probably related to the window manager), I will try to solve this problem but I can't figure it out at the moment. (I don't have the problem with Linux Manjaro or under Windows)

As for the problem of shared settings between the quickmenu and the panel, it's indeed annoying, but I haven't found a way to match the settings between the 2 elements. I need to figure out how to force Freecad to take the settings in the panel.

As for the spacing between the spinbox elements, I can increase the gap by a few pixels, but this is quite dependent on each user's screen resolution, with 5 pixels more it's a little closer to the look of the panel :
[image]

Finally, it's a very good idea to add the possibility of making the spinbox optional. I'll try to work on it, but I've got less time available at the moment, so it may take a while before it's implemented.

Thanks

Grubuntu, 2025-05-04:
TO DO : 
- Solve spinbox's focus issue in some Linux distrib
- Make parameters consistent between spinbox and panel
- Improve spacing of spinbox elements for easier selection
- Add a setting to enable/disable the spinbox

Grubuntu, 2025-05-22:
I've just updated to version 1.12.

I haven't managed to make the settings consistent between the spinbox and the panel (this generates other problems), so I've opted for the solution of disabling the immediate display of the spinbox via an uncheckable box (however, it's still possible to display the spinbox by pressing the PieMenu keyboard shortcut).

I hope this meets your needs.

Just uncheck the box and try it !

[image]

[image]

Thank you for your help and understanding.

closed 2025-05-26 by Grubuntu